tinyvision-ai-inc/pico-ice #40: [REQUEST] USB CDC detect line_coding & Arduino like reboot detection

vanbwodonk, 2024-03-19:
I modified pico_ice_default and pico-ice-sdk/examples/pico_usb_uart with line encoding detection and 1200 bitrate detection for autoreboot. This is very useful if we need a higher bitrate than 115200, and Arduino like auto-reboots when we need to reflash the RP2040. Just need to add some code like this:
```
void tud_cdc_line_coding_cb(uint8_t itf, cdc_line_coding_t const *coding) {
  if (itf == 0) {
    if (coding->bit_rate != baud_rate) {
      baud_rate = coding->bit_rate;
      uart_set_baudrate(uart0, baud_rate);
    }
  }
}

void tud_cdc_line_state_cb(uint8_t itf, bool dtr, bool rts) {
  if (itf == 0) {
    if (rts) {
    }
    if (dtr) {
    } else {
        cdc_line_coding_t coding;
        tud_cdc_get_line_coding(&coding);
        if (coding.bit_rate == 1200) {
          reset_usb_boot(0, 0);
          while (1); 
        }
    }
  }
}
```

Or, if you allow I can make a PR about it.

josuah, 2024-03-19:
Thank you for this! The way it is implemented makes sense to me, and easy to make that configurable with just an `#ifdef` and [a config in `tusb_config.h`](https://pico-ice.tinyvision.ai/group__ice__usb.html).

josuah, 2024-04-17:
I am now back from Embedded World and will be able to switch over to integrating this, likely tomorrow.
Thanks everyone for your patience!

josuah, 2024-04-20:
Adjusting the baud rate works well with what you provided.
I edited it a tiny bit for matching the autoconfig.

Before, baud 1200
[image]

After, baud 4800 (mismatching Pulseview configuration)
[image]

MrJake222, 2024-04-20:
Nice! Does it support resetting and programming with `picotool` (`-f` option)?

josuah, 2024-04-20:
> Does it support resetting and programming with `picotool` (`-f` option)?

Not yet, but https://arduino.github.io/arduino-cli/dev/platform-specification/#1200-bps-bootloader-reset will be supported

josuah, 2024-04-20:
I confirm that after using `picocom -b 1200 /dev/ttyACM0` it is possible to use `picotool` though:

```
$ picotool info
Program Information
 name:      pico_ice_default
 features:  USB stdin / stdout
```

josuah, 2024-04-20:
This landed in v1.6.1, closing :)

closed 2024-04-20 by josuah

MrJake222, 2024-04-20:
Great! I think you only forgot to create new github release with binaries to download

josuah, 2024-04-21:
It was in preparation at that moment, and landed here: https://github.com/tinyvision-ai-inc/pico-ice/releases

MrJake222, 2024-05-11:
> This landed in v1.6.1, closing :)

Did it? Can confirm it works in downloaded `.uf2` but I cannot find any commits or code to build it myself.... Am i blind or you didn't push something?

josuah, 2024-05-11:
Good catch! 

As seen here, `main` was not pushed to `origin/main`, only `develop` to `origin/develop`
```
* 8d0fd06 (HEAD -> develop, tag: v1.6.1, origin/develop, main) implement tud_cdc_line_coding_cb
* 8e92021 (tag: v1.6.0, origin/main, origin/HEAD) stop ICE40 FPGA before initializing SPI flash
```

I pushed it now: https://github.com/tinyvision-ai-inc/pico-ice-sdk/blob/main/src/ice_usb.c#L317