tinyvision-ai-inc/pico-ice #41: Firmware version v1.5.1 can't write to flash (cram works ok)

MrJake222, 2024-04-17:
```
$ dfu-util -l                               
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1209:b1c0] ver=0100, devnum=57, cfg=1, intf=7, path="1-4", alt=1, name="iCE40 DFU (CRAM)", serial="DE6268F24F3D6C21"
Found DFU: [1209:b1c0] ver=0100, devnum=57, cfg=1, intf=7, path="1-4", alt=0, name="iCE40 DFU (Flash)", serial="DE6268F24F3D6C21"
```

```
$ dfu-util -d 1209:b1c0 -a 1 -D hardware.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1209:b1c0
Run-Time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Interface #1 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0101
Device returned transfer size 256
Copying data from PC to DFU device
Download        [=========================] 100%       104090 bytes
Download done.
DFU state(7) = dfuMANIFEST, status(0) = No error condition is present
DFU state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Resetting USB to switch back to runtime mode
Done!
```
```
$ dfu-util -d 1209:b1c0 -a 0 -D hardware.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1209:b1c0
Run-Time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0101
Device returned transfer size 256
Copying data from PC to DFU device
Download        [                         ]   0%            0 bytesdfu-util: Error during download get_status (LIBUSB_ERROR_IO)
```

In version 1.3 flash writing seems to work.

MrJake222, 2024-04-17:
This is fixed by https://github.com/tinyvision-ai-inc/pico-ice-sdk/pull/51

josuah, 2024-04-20:
This was further merged in. Thank you for the detailed report. Will release soon with a few extra other fixes in 1.6.1

closed 2024-04-20 by josuah