tinyvision-ai-inc/pico-ice #44: Is UART1 tested and working?

MrJake222, 2024-05-11:
UART0 is initialized only here: https://github.com/tinyvision-ai-inc/pico-ice/blob/829b9eceeec39da301c08c93d8ef4ce66aaa2dee/Firmware/pico-ice-default/main.c#L120-L122
I can't seem to find equivalent lines for UART1 and the pins should be different.

josuah, 2024-05-11:
Hello!  I think the reason for only UART0 being enabled is that this is the only one piped to the FPGA by default:

https://github.com/tinyvision-ai-inc/pico-ice/blob/829b9eceeec39da301c08c93d8ef4ce66aaa2dee/Firmware/pico-ice-default/tusb_config.h#L59

The several "USB-UART" interfaces presented here: https://pico-ice.tinyvision.ai/md_getting_started.html#autotoc_md6
Are not actually connected to UARTs. I should update the doc to avoid confusion and be more correct.

They are instead bound to different features as described.
In hope I understood your question well, feel free to ask again if I missed something!

MrJake222, 2024-05-11:
I understand that they're only virutal uarts. 

My question is, if i define `ICE_USB_UART1_CDC` to `1`, what will happen? will it work? do i get 2 uarts to fpga?

josuah, 2024-05-11:
> if i define ICE_USB_UART1_CDC to 1, what will happen? will it work? do i get 2 uarts to fpga?

As you guessed, you would not, and modifying the firmware would be necessary.

`pico-ice-sdk` is here to enable to quickly put together a firmware that can be used with the FPGA.
It is aimed to be modified at compile time, i.e. with macros.
It contains examples which acts a bit like reference implementations.

`pico-ice-default` is here to provide most of the SDK features in a ready-to-use way.
I have not thought at bringing configuration for it before.
Would an example in `pico-ice-sdk/examples` with dual UART interfaces enabled work for you?
That would avoid adding code that is disabled by default.
Was there other features that you wanted enabled? Such as the USB-serial to SPI interface?

MrJake222, 2024-05-11:
>  Was there other features that you wanted enabled? Such as the USB-serial to SPI interface?

I've disabled SPI-CDC and enabled UART1