tinyvision-ai-inc/pico-ice #48: pico-ice & pico-ice-sdk

develone, 2024-07-25:
testing pico-ice-sdk & pico-ice 
Below are the steps I am using to test pico-ice-sdk & pico-ice
Trying to follow the steps found in pico-ice/Docs/pico_ice_sdk.md

git clone https://github.com/develone/pico-ice-sdk.git -b test-dev

git clone https://github.com/develone/pico-ice.git -b test-dev

cd pico-ice

cp -r ~/pico-ice-sdk/examples/pico_usb_uart my-new-pico-ice-firmware

cd my-new-pico-ice-firmware

rm -rf pico-ice-sdk pico-sdk

git clone https://github.com/tinyvision-ai-inc/pico-ice-sdk

git clone https://github.com/develone/pico-sdk.git

cd pico-sdk

git submodule update --init

Submodule 'lib/btstack' (https://github.com/bluekitchen/btstack.git) registered for path 'lib/btstack'
Submodule 'lib/cyw43-driver' (https://github.com/georgerobotics/cyw43-driver.git) registered for path 'lib/cyw43-driver'
Submodule 'lib/lwip' (https://github.com/lwip-tcpip/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/mbedtls' (https://github.com/Mbed-TLS/mbedtls.git) registered for path 'lib/mbedtls'
Submodule 'tinyusb' (https://github.com/hathach/tinyusb.git) registered for path 'lib/tinyusb'
Cloning into '/home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/btstack'...
Cloning into '/home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/cyw43-driver'...
Cloning into '/home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/lwip'...
Cloning into '/home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/mbedtls'...
Cloning into '/home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/tinyusb'...
Submodule path 'lib/btstack': checked out '72ef1732c954d938091467961e41f4aa9b976b34'
Submodule path 'lib/cyw43-driver': checked out '8ef38a6d32c54f850bff8f189bdca19ded33792a'
Submodule path 'lib/lwip': checked out '239918ccc173cb2c2a62f41a40fd893f57faf1d6'
Submodule path 'lib/mbedtls': checked out 'a77287f8fa6b76f74984121fdafc8563147435c8'
Submodule path 'lib/tinyusb': checked out '86c416d4c0fb38432460b3e11b08b9de76941bf5'

cd ../

mkdir build

cd build

devel@pi5-80:~/pico-ice/my-new-pico-ice-firmware/build $ cmake ..
PICO_SDK_PATH is /home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3 (found version "3.11.2") found components: Interpreter 
TinyUSB available at /home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/btstack
cyw43-driver available at /home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/lwip
mbedtls available at /home/devel/pico-ice/my-new-pico-ice-firmware/pico-sdk/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/devel/pico-ice/my-new-pico-ice-firmware/build

make

[ 74%] Building C object CMakeFiles/pico_usb_uart.dir/pico-sdk/src/rp2_common/pico_stdio_uart/stdio_uart.c.obj
[ 74%] Building C object CMakeFiles/pico_usb_uart.dir/pico-ice-sdk/src/ice_usb.c.obj
/home/devel/pico-ice/my-new-pico-ice-firmware/pico-ice-sdk/src/ice_usb.c:49:10: fatal error: uf2.h: No such file or directory
   49 | #include "uf2.h"
      |          ^~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/pico_usb_uart.dir/build.make:877: CMakeFiles/pico_usb_uart.dir/pico-ice-sdk/src/ice_usb.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1528: CMakeFiles/pico_usb_uart.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

develone, 2024-07-25:
tested on Raspberry Pi5 running Raspberry OS 64 bit.
Also tested on Windows 10 in Windows Subsystem for Linux WSL.
Need to be able to compile the pico-ice-sdk before purchasing a pico-ice board.

develone, 2024-07-26:
devel@pi5-80:~/pico-ice-sdk/examples/ice_makefile_blinky $ export OSS_CAD_SUITE=/usr/local
devel@pi5-80:~/pico-ice-sdk/examples/ice_makefile_blinky $ make -f Makefile 
/usr/local/bin/yosys -q -p "read_verilog -sv top.sv; synth_ice40 -top top -json gateware.json"
/usr/local/bin/nextpnr-ice40 -q --randomize-seed --up5k --package sg48 --pcf ice40.pcf --json gateware.json --asc gateware.asc
/usr/local/bin/icepack gateware.asc gateware.bin

josuah, 2024-07-26:
Hello! It looks like this requires some `git submodule update --init` inside the `pico-ice-sdk` repo.
Could you try this and confirm it still has the same error?

josuah, 2024-07-26:
This is where the tinyUF2 library is expected to be, and it contains `uf2.h`:
https://github.com/develone/pico-ice-sdk/tree/main/lib

josuah, 2024-07-26:
Looking again, I cannot find the `pico_ice_sdk.md` file. Maybe it is out of date?
I think this latest doc might work better for you: https://pico-ice.tinyvision.ai/md_getting_started.html

develone, 2024-07-28:
Was this code developed in Micosoft Windows? Could this be the issue?

develone, 2024-07-28:
uf2.h & board_api have comments to Microsoft. If you add uf2.h to pico-ice-sdk/include you get the missing board_api.h.

josuah, 2024-07-28:
Sorry I had limited time 2 days ago.

> Was this code developed in Microsoft Windows?

It was developed under Linux (Alpine) and tested on Linux (Ubuntu) on WSL, MinGW from Yosys (for the RTL part at least), and Mac OS by users.

> uf2.h & board_api have comments to Microsoft

UF2 is a format invented by Microsoft to work around the fact that custom USB devices are difficult to use for users, as it requires manually selecting a driver such as WinUSB or libusbK using tools like Zadig.
For this reason, Microsoft developed the UF2 format that permits an USB device to be recognized as a storage device with a FAT32 file system, accepting files uploads of files in the UF2 format. https://github.com/microsoft/uf2

It is a bit of a hack, but permits to do firmware upgrade without installing drivers manually on Windows, and without changing permission issues on most systems. Only a drag-and-drop of a file in the UF2 format.

This is one of the several upload protocols supported by the pico-ice, along with DFU (the standard defined by USB.org which requires some workarounds on Windows but works well via the `dfu-utils` on Linux/MacOSX):
https://pico-ice.tinyvision.ai/md_programming_the_fpga.html

In order to support it for the pico-ice, we picked the TinyUF2 library by Adafruit:
https://github.com/adafruit/tinyuf2

This needs to be populated in this directory in order to provide all the UF2 implementation used for supporting firmware upload through drag-and-drop:
https://github.com/tinyvision-ai-inc/pico-ice-sdk/tree/main/lib

## TL;DR:

Which means that if you run `ls pico-ice-sdk/lib/tinyuf2/`, you should see this:
```
$ ls pico-ice-sdk/lib/tinyuf2/
LICENSE  README.md  apps  changelog.md  cmake  lib  ports  src
```

If you do not, this means a submodule is missing. You might have outdated documentation, and can start over from instructions here: https://pico-ice.tinyvision.ai/md_getting_started.html#autotoc_md8

Or only try this line to fix your ongoing directory.

```
git -C pico-ice-sdk submodule update --init
```

If you have all some files listed as above, this means that maybe something else went wrong. You might need to delete the `build` directory and start over.

develone, 2024-07-29:
Hello! It looks like this requires some git submodule update --init inside the pico-ice-sdk repo.
This appears to be the fix.  
steps found https://github.com/develone/pico-ice/blob/test-dev/myDocs/notes-pico-ice-2.txt
All the files in build folder https://github.com/develone/pico-ice/blob/test-dev/myDocs/built-files.txt
Thanks now I can purchase the pico-ice knowing that my learning curve will be reduce.
Please take a look at this pdf.  Adding a file system and FreeRtos might add some enhanced features.

develone, 2024-07-29:
I have been working with the ice40 HX8K using zipcpu. 
https://github.com/develone/rp2040-freertos-project/blob/dev/doc/pico-test-read.pdf
https://github.com/develone/102121icozip/blob/catzip/doc/dwt3lvls050924.png

josuah, 2024-07-29:
Glad it ended-up working! That is an interesting project you have. A good improvement over https://github.com/mcknly/breadboard-os

Note that `pico-ice/Docs/pico_ice_sdk.md` disappeared 7 month ago: the documentation got updated.
You will be able to get the latest documentation by doing a `git pull` inside `pico-ice`.

develone, 2024-07-29:
Where is the best place to order a pico-ice board tindie or the tinyvision store?

develone, 2024-07-29:
I ordered 2 pico-ice boards Order #484119 Receipt at Tindie.   I might be looking to some more for students at UTEP here in El Paso, Texas.  Do you have pricing for students.

josuah, 2024-07-29:
Hello and sorry for the delay. Here is the recommendation from who runs tinyVision.ai:
- Lectron-Z for EU and UK mostly (helps with taxes)
- Tindie/tinyVision.ai for US mostly (ships from here)
- Elecrow for all (including EU/UK/US, ships internationally).
I will ask about student/bulk pricing and let you know!

josuah, 2024-07-29:
For this you would need to reach out via one of these ways:
https://tinyvision.ai/pages/contact-us
https://discord.gg/t2CzbAYeD2

josuah, 2024-07-29:
I am closing the issue as the original problem was solved, but no problem with discussing more!

closed 2024-07-29 by josuah